opensolarproject/OSPController #2: Connection of ESP32 unclear

labroid, 2019-11-05:
Great project! Unfortunately, I was unable to gather how the ESP32 is connected to the converter. You mention for labeled pins on the converter, but not how they connect to the ESP32. It would be great to have a sentence on that.

While I'm here, I am an EE with power conversion experience and I agree this is an MPPT despite comments from others. It seems you achieve MPPT because you are controlling for input peak, and output regulation suffers - which is a fine approach if the load can tolerate it. I'd like to throw my support into the hack-a-day comments and first would like to better understand the connection to the converter, and also to confirm what you are controlling (by the code it seems you are changing the load current) Is there an API to the converter UART you can point us to?  Thanks!

labroid, 2019-11-05:
Saw your Amazon comment and found the control library. To make it easy for others:  https://benjames.io/2018/06/29/secret-uart-on-chinese-dcdc-converters/ is the article, and the library is at https://github.com/bengineer19/BuckPSU

t413, 2019-11-15:
Nice job finding the links there. 😊 You can also find it in the [about page in the wiki](https://github.com/opensolarproject/OSPController/wiki/About#supplying-the-power).

I had some issues with that library and wrote my own, it's pretty simple and modular if you'd like to use it elsewhere. Here's where it is [in the code](https://github.com/opensolarproject/OSPController/blob/master/lib/MPPTLib/utils.h).

labroid, 2019-11-15:
Excellent. Many thanks!

mrstuff, 2020-02-20:
Chagrin to admit my board difference will mean I get to try translate what's going on here, and I take it that I am supposed to get all I need to know tracking down and connecting my nodemcu esp32s to the appropriate pins? from the buck converter library ok but I  would really love a down to the pin diagram of what you have going on there, especially since I'm having a heck of a time connecting to my esp32s and have yet to make any headway.
Much appreciated

austrisv, 2020-05-25:
On my esp32 devkitc this did work
|psu|esp32|
|R|16|
|T|17|
|G|gnd|
|V|not connected|

mrstuff, 2020-05-27:
thanks much !