opensolarproject/OSPController #6: Different panel voltage

03K, 2020-01-16:
I have all the parts ready for solderong :) However, my input voltage is 48V. I will drop down to 42V in order to mainly charge electric skateboards :) Can you point out where I need to tweak your setup in order to work with approx. 48V. I see the voltage divider must be different, I am thinking to put small dc-dc converter there too so I can adjust if I change my panels condiguration. Cirrently its 4 panels in series. Do I need to modify the software?

austrisv, 2020-05-26:
if you need a different output voltage - you can set it from drok side or via serial/curl interface (outvolt). 
Though my understanding is that for charging you don't need to do it. As you will connect the skateboard/battery - the voltage will 'automagically' drop to the batteries level. And esp32 will keep track of current (ampers) delivered to the battery. At least that was my observation (which I also described/asked to confirm  in a question https://github.com/opensolarproject/OSPController/issues/12)

t413, 2020-06-22:
Hey @03K, it was great chatting with you on the [discord chat](https://discord.gg/WStK4b) and getting you ready to build when you're back home with your gear. 

For anyone else wondering: @03K has 4x 13V<sub>disconnected</sub> = 52V max. Sounds like The V<sub>mppt</sub> of his panels together may be ~48V. This is pretty nonstandard, usually panels are sold with a nominal voltage of 12, 18, 20, or 24V. See the [table on this page](https://www.altestore.com/howto/solar-panels-pv-and-voltages-a98/) for more. The goal is to make sure your _VMP - typical_ (that's the approximate voltage at maximum-power-point _typically_) is **above** your max battery voltage. That's because we're using a buck converter that can only regulate voltage down. 

The DROK regulator we're using can take an open-voltage of ~80V (mine is 82V and works fine) so it's almost always better to get another panel in series to have a higher input voltage.

closed 2020-06-22 by t413