opensolarproject/OSPController #7: PIN CHANGE  Photography not Incongruent ambiguity!

mrstuff, 2020-02-22:
I offer Photography (hopefully) not Incongruent ambiguity!
          though I see the latest code reflects the updated board platform.io data, it seems that while a glance at the code would conform to the simple enough principal at hand, and its not like you can go hold everybody's hand and patently walk thru the nitty gritty, right? So!
That's where I come in, so I see the instructions are well written and concise is good, please! Brevity

 but I see the pin change from 36 to 32 is that right?
Its must be since no such pin exists on the nodemcu esp32 unless your talking about the esp32-wroom and i see the photo on the roof depicts a close shave with some hairy looking jumper nest work and I would just love to know exactly what board and pins we were looking at, tho its not like I couldn't meet the RX tx and involt requirement o

I think your page layout and accounts for the right amount of facts and does a moderate to fair job of conveying in no uncertain terms the often detailed if sometimes incomplete array of materials and pertaining operations, however I would really like to see some fine to the point labeling and description. 

I have been struggling with my understanding of exactly how and where everything is all connected. I understand this is git not instructables or hackaday and the focus is on the code over often erroneous explanation. 

Hey this is a really neat build and I gotta say for what you've got its my delight if I might take photos and or draw a pin for pin schematic..
How ever you see fit to share, much appreciated and I would prefer my critical, poorly informed and exceedingly inexperienced notes be viewed as my vehement will to contribute effectively.

t413, 2020-05-13:
Great to hear you’re working on a system! Sorry for not getting the documentation up to snuff to get you going easily. 

What board did you get? The one I linked or another esp32 board? Send me a link and I’ll Try and verify pins. 

It really sounds like I need to make the pins be user-selectable ASAP.

mrstuff, 2020-05-13:
I got the board listed but fried it somehow but not before getting the pins right I think and i even got it up and running for a day before I foolishly blew a fuse on the drok and with it the esp32s so my bad for sure. 
So delighted am I to have reached you that I am doubling down my efforts to reproduce the working setup but I still have a hankering for a neat pin to pin labeled unambiguous as heck, could perhaps just add some tags to your existing photo if you see fit
A thousand times thank you

t413, 2020-05-13:
Not your bad. My lack of follow through left you without help and guidance or at least good documentation. 

Can I order you a new esp32? 

The trouble with the pins is how un-transparent they are and how much they subtly change mapping between boards or just don’t exist at all. It always seems like the pins that are broken-out most & most-consistently on boards are all actually needed by WiFi :| 

Making things user-configurable and easily testable would be a far better way. Let’s see here.. (grabs test setup)

mrstuff, 2020-05-13:
I must say I did have a bit of trouble when it came time to connect the wifi and whatnot thru the serial port ; i think I had gotten the pins right then and made it as far as successfully visiting the local mqqt page but without any readable data so I understand that's not enough info to really do anything with but I thought it worthy of mention

mrstuff, 2020-05-20:
A search for the integer 32 showed word for word (sorry)

27 | void doSweepStep();
-- | --
28 | bool hasCollapsed() const;
29 | int getCollapses() const;
30 |  
31 | uint8_t pinInvolt_ = 32;
32 | float inVolt_ = 0, wh_ = 0;
… |  
34 | double currentCap_ = 8.5;
35 | CircularArray<uint32_t, 32> collapses_;
36 | int measperiod_ = 200, printPeriod_ = 1000, psuperiod_ = 2000;

So what I want to know is what pinInvolt is on line 31

austrisv, 2020-05-25:
It took me a while to undeestamd line 31 :)

It is the esp32 pin 32 to which you should connect the voltage divider's output.
[image]

t413, 2020-06-22:
Hey @mrstuff! 

I've been working with @austrisv on the project and getting it up to par. Are you still interested in getting it running? I made a [discord channel](https://discord.gg/WStK4b) that's been really nice and active. Come join!

mrstuff, 2020-06-22:
thanks for reaching out and i am much interested and honored to be a part
ot this thang... i do have some real concerns about kind of specific things
and am looking forward to more a more funtional articulation of my ideas
and hope to ask some neat questions too
-Hans
 --you da man

closed 2020-06-22 by mrstuff