opensolarproject/OSPController #8: Q - is it also battery charge controller

austrisv, 2020-04-27:
Dear Tim, Thank You for time and efforts putting together the design, documentatiom and sharing the code. Much appreciated! I'm a newbie in the Solar business, sorry if stupid question. Do I understand it right that your OSPController takes care of a charge current and voltage, but does not otherwise monitor the health of battery - e.g. shutting off the appliances if battery discharged to low and panel not generating anything (night).
I'm planning solar power for my shed, including battery for evemings and a little ac inverter. I do like your approach a lot. Trying to understand what (if anything) I need in addition for safe battery operation. Thank You!

mrstuff, 2020-05-13:
I too have this same question and would really appreciate some, any feedback alas this may never happen so I can only suggest that in the code and in the buck psu limited documentation the answer could be found. Also doesn't anyone who isn't charging a 94v wheel battery have to adjust the code so it floats at the proper range for each setup? 
might start by asking the guy who posted the buck psu library on github since our illustrious creator has gone incommunicado

t413, 2020-05-13:
Sorry! I never realized I had folks working on this! GitHub emails never made it to my inbox :/

What kind of battery are you thinking about / working with? I just haven’t coded-in a float phase for lead acid yet but it should still work fine. Most anything else should work great. 

You’re right, I don’t have any load-side low-voltage cutoff. You’ll need a relay or a big mosfet (relay might be a better idea in most cases). I’ll add that to the roadmap for something to add-in.. but for now, and to be extra sure you don’t kill your batteries, I’d recommend a cheap external low-voltage-cutoff. Like [this](https://www.amazon.com/Digital-Battery-Low-Voltage-Protection/dp/B07929Y5SZ/)

austrisv, 2020-05-14:
Thank You @t413 a million, you are our rock star!:)
I had tha buch module order anyways, hoping will figure it out somehow. As well as thinking - if using just mains inverter - it shall have the cutoff built in anyways.
And having and extra voltage divider for checking battery line plus relay was my plan B (and then can maybe still manage to send last mqtt message before cutting off). Though, the external cutoff looks nice, too. 
On battery type - to begin with, will see what I can find available in friends garage. Most likely starting with lead acid. Have a grip of how system works and how much I lack and then decide (in my case it's off-grid garden shed; not too demanding; more a curiosity project/want to give solar power a try :))

happytm, 2020-05-14:
@austrisv ,

Please  keep us posted on your progress. There must be lot of curious minds.

Thanks.

austrisv, 2020-05-26:
This is my quick&dirty (particularly the battery) POC-balcony setup 
[image]
[image]
[image]
Missing:
* fuses/circuit breakers
* battery protection
* capacitors
* larger fun
* wifi hotspot

happytm, 2020-05-26:
@austrisv 

Thank you for sharing clear images. I am sure it will be helpful to many people like me.

mrstuff, 2020-05-27:
horray! thanks so much awesome!

t413, 2020-06-11:
Thanks for the diagram and photos! I really love seeing a working setup! 

I'm closing since this was a question and it seems like it's been happily answered. 

And.. I just created a discord channel! [discord.gg/MRQvKR](https://discord.gg/MRQvKR)

closed 2020-06-11 by t413