opensolarproject/OSPController #9: PlatformIO defects ... err say whaa?

mrstuff, 2020-05-21:
cppcheck | LOW | STYLE | CWE-398:Class 'PowerSupply' has a constructor with 1 argument that is not explicit. | …1/lib/MPPTLib/utils.h:12:5
-- | -- | -- | -- | --
cppcheck | LOW | STYLE | CWE-398:Class 'PubSubClient' has a constructor with 1 argument that is not explicit. | …src/PubSubClient.h:117:4
cppcheck | LOW | STYLE | CWE-0:The function 'write' overrides a function in a base class but is not marked with a 'override' specifier. | …src/PubSubClient.h:170:19
cppcheck | LOW | STYLE | CWE-0:The function 'write' overrides a function in a base class but is not marked with a 'override' specifier. | …src/PubSubClient.h:173:19

Translate???

mrstuff, 2020-05-27:
I'm thinking I've just got something wrong on my end and it's a minor defect I expect

t413, 2020-06-11:
These are just warnings, specifically style warnings somehow. 3/4 are in the MQTT library PubSubClient, the one is a strange one about my PowerSupply class that, yeah, makes no sense that it'd be even a style warning. Odd. 

Ignore! Also how did you generate these?

closed 2020-06-11 by t413