opensolarproject/OSPController #10: wifi does not connect to router

austrisv, 2020-05-24:
This is the console output after fresh upload from vsc, sending wifiap and wifipass and doing save and restart
```startup, ID 3E334FC4 41D7
loaded key PSUperiod to 2000
loaded key autostart to false
loaded key autosweep to 600
loaded key currentcap to 8.500
loaded key measperiod to 200
loaded key mqttFeed to solar/mppt
loaded key mqttPass to 
loaded key mqttServ to 
loaded key mqttUser to 
loaded key pgain to 0.005
loaded key printperiod to 1000
loaded key pubperiod to 1000
loaded key ramplimit to 2.000
loaded key setpoint to 0.000
loaded key vadjust to 116.50
loaded key wifiap to aaaaaa
loaded key wifipass to 12345678
finished setup
6.9Vin -> 0.00Wh <0.00V out 0.00A 0en>
wifi event
wifi event
0.0Vin -> 0.00Wh <0.00V out 0.00A 0en> 
0.0Vin -> 0.00Wh <0.00V out 0.00A 0en> 
wifi event
can't pub connect, wifi 0 pub 0
can't pub connect, wifi 0 pub 0
0.0Vin -> 0.00Wh <0.00V out 0.00A 0en> [pub disconnected] 
wifi event
psu update fail
0.0Vin -> 0.00Wh <0.00V out 0.00A 0en>
0.0Vin -> 0.00Wh <0.00V out 0.00A 0en> 
can't pub connect, wifi 0 pub 0
0.0Vin -> 0.00Wh <0.00V out 0.00A 0en> [pub disconnected] 
0.0Vin -> 0.00Wh <0.00V out 0.00A 0en> 
wifi event
psu update fail
0.0Vin -> 0.00Wh <0.00V out 0.00A 0en>
can't pub connect, wifi 0 pub 0
0.0Vin -> 0.00Wh <0.00V out 0.00A 0en> [pub disconnected] 
0.0Vin -> 0.00Wh <0.00V out 0.00A 0en> 
wifi event```

austrisv, 2020-05-24:
In solar.cpp doConnect I added debug output
``` Serial.println("doSetup 1 " + wifiap + " " +wifipass);
wight after the WiFi.begin and was surprised to see pass printed first, followed by AP.
Then I 'hardcoded'
``` WiFi.begin("aaaaaa", "12345678")
and voila, it is connecting

austrisv, 2020-05-24:
I must be doing smtg wrong ... as it was also not connecting to MQTT. Did the same trick here
```db_.client.connect("MPPT", "myuser", "mypass");
and is publishing

mrstuff, 2020-05-27:
i also experienced identical behaviour and am looking forward to emulating you and you have done here and would be happy to try speak up if and when these little foibles can be fleshed out.
I have a concern for the scale of my set up ;dwarfs this and I am using deep cycle marine batteries connected in parallel . I wilk post the numbers  and details in a new issue I think

mrstuff, 2020-06-08:
Would precisely where to insert your additional lines of code or even a modified working repo be too great a task to illustrate explicitly?

austrisv, 2020-06-09:
> Would precisely where to insert your additional lines of code or even a modified working repo be too great a task to illustrate explicitly?

I have not made modification to the original repo as I did not had chance to properly debug and understand what was wrong.
What I did, after several attempts to save wifiap and  wifipass over serial, I changed this line:
https://github.com/opensolarproject/OSPController/blob/528c94498cceb0c658f1b281f4035ed74c668f38/lib/MPPTLib/solar.cpp#L92
to
```WiFi.begin("my_ssid, "my_password");
**N!B!** be careful as it will not work like that on erased flash/fresh upload .... the previous check for  wifiap.length and wifipass.length will fail. So, you will have to store something in wifiap and wifipass and then the code will pick up hardcoded values.

And I had to do the same for MQTT connection as well
https://github.com/opensolarproject/OSPController/blob/528c94498cceb0c658f1b281f4035ed74c668f38/lib/MPPTLib/solar.cpp#L108-L109 

Does not sound right to me, but without that was not working :(

austrisv, 2020-06-09:
@mrstuff , have you managed to get through the voltage divider jungles?

> now I've only to muster the cajones to attach the voltage divider

austrisv, 2020-06-09:
OK, I guess I figured out what was wrong. As I was using Arduino's serial monitor to save wifiap and wifipass - it had the default line ending "both nl&cr". As I changed to "No line ending" - setting the AP and Pass worked fine with original code. Makes sense:) Thus closing (canceling) this issue as I was original reporter of problem,
[image]

closed 2020-06-09 by austrisv