opensolarproject/OSPController #12: Q - is it the correct behavior of the system?

austrisv, 2020-05-26:
- [ I was surprised to notice that converter's lcd comes powered on even if only connect battery to the output (and input is empty (or panel covered with blanket). ]

- [Attaching MQTT Explorer screenshot with few comments With everything connected, as soon as I turn on the converter (click OK button), the involt drops to about same level as battery voltage, which is also equal to outvolt.
If I disconnect the battery - the involt jumps up to 18-20V what's measured at solarpanel output. And the outvolt jumps to the voltage value set on converter (I had it manually set to 14V).

Can't make my head around - is it the expected voltage drop = equalize to the batteries level at both sides. Or little drop at input would have been expected, but not to the battery level?]

austrisv, 2020-05-26:
judging by the quote from here
https://www.solar4rvs.com.au/buying/buyer-guides/choosing-the-right-solar-charge-controller-regulat/ 

> Note that when the switch is OFF the panel voltage will be at the open circuit voltage (Voc) and when the switch is ON the panel voltage will be at the battery voltage + voltage drops between the panel and the controller.

my 2nd bullet is the expected behavior of system.

t413, 2020-06-11:
Hmm. I'd like to get a little more information, specifically from the status messages sent out on the serial port. Can you have your ESP32 plugged into your computer and use the serial monitor to collect some output? 

What panel are you using? Specifically what's its Vmp? Looks like you're using a lead acid battery?

I didn't even notice your issues until today, I really apologize. I can see that github issues aren't the best at debugging this stuff.. So let's try discord and see how that works. Just created a channel [discord.gg/MRQvKR](https://discord.gg/MRQvKR), come message me.

austrisv, 2020-06-11:
My panel is this one
https://a.aliexpress.com/_dUM7PiY
Vmp 18V and lead acid battery (for time beeing). Varta M18

austrisv, 2020-06-12:
Here is my log output (publishing here, in case if in future anyone interested)
```load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
startup, ID 3E334FC4 41D7
loaded key PSUperiod to 2000
loaded key autostart to false
loaded key autosweep to 600
loaded key currentcap to 8.500
loaded key measperiod to 200
loaded key mqttFeed to solar/mppt
loaded key mqttPass to susuri
loaded key mqttServ to 78.84.150.35
loaded key mqttUser to openhabian
loaded key pgain to 0.005
loaded key printperiod to 5000
loaded key pubperiod to 30000
loaded key ramplimit to 2.000
loaded key setpoint to 0.000
loaded key vadjust to 27.20
loaded key wifiap to XXXXXXXX
loaded key wifipass to ########
finished setup
wifi event
wifi event
wifi event
wifi event
Wifi connected! hostname: mpptESP-C4
IP: ########
Connecting MQTT to openhabian@########
PubSub connect success! 
got sub value solar/mppt/prefs/PSUperiod -> 2000
got sub value solar/mppt/prefs/autostart -> false
got sub value solar/mppt/prefs/autosweep -> 600
got sub value solar/mppt/prefs/currentcap -> 8.500
got sub value solar/mppt/prefs/measperiod -> 200
got sub value solar/mppt/prefs/pgain -> 0.005
got sub value solar/mppt/prefs/printperiod -> 5000
got sub value solar/mppt/prefs/pubperiod -> 30000
got sub value solar/mppt/prefs/ramplimit -> 2.000
got sub value solar/mppt/prefs/setpoint -> 0.000
got sub value solar/mppt/prefs/vadjust -> 27.20
got sub value solar/mppt/wh -> 368.49
restored wh value to 368.49
got sub value solar/mppt/prefs/PSUperiod -> 2000
got sub value solar/mppt/prefs/autostart -> false
got sub value solar/mppt/prefs/autosweep -> 600
got sub value solar/mppt/prefs/currentcap -> 8.500
got sub value solar/mppt/prefs/measperiod -> 200
got sub value solar/mppt/prefs/pgain -> 0.005
got sub value solar/mppt/prefs/printperiod -> 5000
got sub value solar/mppt/prefs/pubperiod -> 30000
got sub value solar/mppt/prefs/ramplimit -> 2.000
got sub value solar/mppt/prefs/setpoint -> 0.000
got sub value solar/mppt/prefs/vadjust -> 27.20
12.7Vin -> 368.49Wh <0.00V out 0.00A 0en> [published 12] 
12.7Vin -> 368.52Wh <12.68V out 1.70A 1en> 
12.7Vin -> 368.55Wh <12.68V out 1.70A 1en> 
12.8Vin -> 368.58Wh <12.68V out 1.72A 1en> 
12.7Vin -> 368.61Wh <12.68V out 1.73A 1en> 
12.7Vin -> 368.64Wh <12.69V out 1.75A 1en> 
12.8Vin -> 368.67Wh <12.68V out 1.77A 1en> [published 6] 
```

austrisv, 2020-06-12:
Here is my log output (publishing here, in case if in future anyone interested)
```load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
startup, ID 3E334FC4 41D7
loaded key PSUperiod to 2000
loaded key autostart to false
loaded key autosweep to 600
loaded key currentcap to 8.500
loaded key measperiod to 200
loaded key mqttFeed to solar/mppt
loaded key mqttPass to aaaaaa
loaded key mqttServ to ########
loaded key mqttUser to aaaaaa
loaded key pgain to 0.005
loaded key printperiod to 5000
loaded key pubperiod to 30000
loaded key ramplimit to 2.000
loaded key setpoint to 0.000
loaded key vadjust to 27.20
loaded key wifiap to XXXXXXXX
loaded key wifipass to ########
finished setup
wifi event
wifi event
wifi event
wifi event
Wifi connected! hostname: mpptESP-C4
IP: ########
Connecting MQTT to openhabian@########
PubSub connect success! 
got sub value solar/mppt/prefs/PSUperiod -> 2000
got sub value solar/mppt/prefs/autostart -> false
got sub value solar/mppt/prefs/autosweep -> 600
got sub value solar/mppt/prefs/currentcap -> 8.500
got sub value solar/mppt/prefs/measperiod -> 200
got sub value solar/mppt/prefs/pgain -> 0.005
got sub value solar/mppt/prefs/printperiod -> 5000
got sub value solar/mppt/prefs/pubperiod -> 30000
got sub value solar/mppt/prefs/ramplimit -> 2.000
got sub value solar/mppt/prefs/setpoint -> 0.000
got sub value solar/mppt/prefs/vadjust -> 27.20
got sub value solar/mppt/wh -> 368.49
restored wh value to 368.49
got sub value solar/mppt/prefs/PSUperiod -> 2000
got sub value solar/mppt/prefs/autostart -> false
got sub value solar/mppt/prefs/autosweep -> 600
got sub value solar/mppt/prefs/currentcap -> 8.500
got sub value solar/mppt/prefs/measperiod -> 200
got sub value solar/mppt/prefs/pgain -> 0.005
got sub value solar/mppt/prefs/printperiod -> 5000
got sub value solar/mppt/prefs/pubperiod -> 30000
got sub value solar/mppt/prefs/ramplimit -> 2.000
got sub value solar/mppt/prefs/setpoint -> 0.000
got sub value solar/mppt/prefs/vadjust -> 27.20
12.7Vin -> 368.49Wh <0.00V out 0.00A 0en> [published 12] 
12.7Vin -> 368.52Wh <12.68V out 1.70A 1en> 
12.7Vin -> 368.55Wh <12.68V out 1.70A 1en> 
12.8Vin -> 368.58Wh <12.68V out 1.72A 1en> 
12.7Vin -> 368.61Wh <12.68V out 1.73A 1en> 
12.7Vin -> 368.64Wh <12.69V out 1.75A 1en> 
12.8Vin -> 368.67Wh <12.68V out 1.77A 1en> [published 6] 
```

austrisv, 2020-06-15:
Answering the original question of myself - apparently not 100% expected behavior. DROK beeing powered from output side - yes, confirmed with Tim - the same happens on his device, too. But the mppt action was not kicking on. This is how the log will look like when manually sending "sweep" command over serial.
```19.4Vin -> 573.97Wh <12.59V out 4.67A 1en> 
19.5Vin -> 574.05Wh <12.59V out 4.67A 1en> 
19.5Vin -> 574.13Wh <12.59V out 4.67A 1en> 
SWEEP START c=0.000, (setpoint was 0.000)
<sweep > starting sweep
19.5Vin -> 574.22Wh <12.60V out 4.67A 1en> [adjusting 0.00A (from 0.02A)] 
23.2Vin -> 574.22Wh <12.60V out 0.87A 1en> [adjusting 0.00A (from 0.04A)] 
23.1Vin -> 574.22Wh <12.60V out 0.02A 1en> [adjusting 0.00A (from 0.06A)] 
23.1Vin -> 574.22Wh <12.60V out 0.04A 1en> [adjusting 0.00A (from 0.08A)] 
23.1Vin -> 574.22Wh <12.60V out 0.07A 1en> [adjusting -0.00A (from 0.10A)] 
23.1Vin -> 574.22Wh <12.60V out 0.10A 1en> [adjusting -0.00A (from 0.12A)] 
23.1Vin -> 574.22Wh <12.60V out 0.12A 1en> [adjusting 0.00A (from 0.14A)] 
23.1Vin -> 574.22Wh <12.60V out 0.13A 1en> [adjusting 0.00A (from 0.16A)] 
23.1Vin -> 574.22Wh <12.60V out 0.15A 1en> [adjusting 0.00A (from 0.18A)] 
...
6.6Vin -> 575.25Wh <12.62V out 5.96A 1en> [adjusting 0.00A (from 6.00A)] 
12.8Vin -> 575.25Wh <12.62V out 5.72A 1en> [adjusting 0.00A (from 6.02A)] 
12.8Vin -> 575.25Wh <12.62V out 5.72A 1en> 
SWEEP DONE. c=5.820 v=17.488, (setpoint was 0.000)
```

closed 2020-06-15 by austrisv

t413, 2020-06-15:
YES. That’s it finding the best possible input voltage, 17.488V, and then it should start running according to that. That’s the maximum power point’s voltage at that moment. 

This is all excellent  feedback for how to make the system perform better for everyone right off the bat without any work. I’ll be removing ‘autostart’ and adding in more intelligence instead.

t413, 2020-06-15:
YESSSSSSS that second graph is perfect! Note the perfect arc of the day’s sun! And a wonderful illustration between a direct solar-to-battery connection vs running with MPTT. Sorry it wasn’t running for you right off the bat. Needs to be easier.