opensolarproject/OSPController #13: IF block missing thus getting false "PubSub connect ERROR!"

austrisv, 2020-06-09:
Looks to me line 109 missing "{" at the end and line 115 missing "}" at the beginning. 
https://github.com/opensolarproject/OSPController/blob/528c94498cceb0c658f1b281f4035ed74c668f38/lib/MPPTLib/solar.cpp#L109-L115

Asis it is giving the connect error message even in case of successful connect.

austrisv, 2020-06-09:
p.s.
@t413, I would have submitted a pull request, but either I do not have permission ... or something not working with my setup

t413, 2020-06-11:
Hey Austrisv, thanks for the issue and trying to add a fix!

I went ahead and added a fix.. and fixed the other two compiler warnings I had as well with 5d157bf18bca80fcc96b262c56fac7f062d10ff2. 

When you want to do a pull request first you have to use the fork button to create a copy of the repository under your user (it'll be github.com/austrisv/OSPController), and push your code to there. That'll let you organize all the changes you want until you go and create a pull request specifically from your fork's page. Here's a [little howto](https://opensource.com/article/19/7/create-pull-request-github).

closed 2020-06-11 by t413