opensolarproject/OSPController #15: Improvement: put it asleep during night for 24/7 operation

austrisv, 2020-06-15:
I wonder, would it make sense to stop actively working with PSU after the sunset and before sunrise?
May still checking&reporting battery voltage (outV) at some interval, but deepsleeping the rest of the time?

If doing so, I guess, it should request sun-triggers based on configured latitude/longitude (e.g. in a similar way how Tasmota does it)

t413, 2020-06-17:
Just sleeping for 5 minutes at a time would reduce power consumption by a whole lot and still allow for nice reporting of battery voltage.

Definitely a good TODO.

rin67630, 2021-02-16:
Your far bigger power hog at night is the DROK/DPS. 
I suppose, the DROK has no enable input? The DPS don't.
So the only way would have been an ideal diode between the converter and the battery, so the battery will not back-feed the converter permanently.
You then lose the battery voltage reporting or you additionally need another analogue input to measure directly the battery voltage beside the buck module.

porteusconf, 2024-11-12:
In 2024, seems 20mA is about what most converters need with no load, both for LCD or even OLED units! So, does that make the DROK/DPS a power hog? ;-) Details below...

A 2024 review for DROK with LCD said "Power draw (11.71V in) was 19mA with backlight off and in sleep (off) mode, 45mA with backlight on maximum."  [for newer "DROK LCD Display Buck Boost Converter, DC 6-36V to 0.5-36V 4.5A 40W Step Up Step Down Converter, 5V 12V 24V DC Power Supply, Adjustable Voltage Regulator Step Up Down Transformer for Solar Panels RV" ](https://www.amazon.com/gp/customer-reviews/R25QW47JJAKUSG/ref=cm_cr_srp_d_rvw_ttl?ie=UTF8&ASIN=B0D66XRHNW)  

I have a [SK160 with color OLED display instead of LCD ](https://www.amazon.co.uk/Yssevlon-Converter-Adjustable-Regulated-Constant/dp/B0DDPZ1L2R). The brightness can be set from 1 to 5, but I don't see any way to turn off display or sleep. Even so, at lowest brightness of 1 (least bright), the current is only 21mA, surprisingly only 2ma more than the LCD model above with backlight off.

I thought I saw some other units with power-draw of only 10mA or so, but I cannot find them at the moment. I too want to minimize power drain if I can. As not sure I trust the DROK, I bought a separate low-voltage cut-out circuit with a mechanical relay I was thinking of using between battery and the converter, but have not measured power-draw of that yet. I suspect it may need enough power to keep the relay energized, so that might be a worse than the 10-20mA of converter. I guess I'll have to test the DROK to see if the Low-Voltage cut-off works reliably, and if so, maybe leave out the separate relay circuit.