opensolarproject/OSPController #18: battery not charging

schmurfy, 2020-11-11:
Hello, I already had some esp32 lying around so I bought the power supply to build myself a small scale solar system for my balcony but I can't figure how to make this works.

I have my esp32 flashed with its input connected to a 12v power supply (for testing) and the output connected to an almost full 12v lead acid battery, here are the values reported on the web interface:
```json
{
  "collapses":"0",
  "currFilt":"0.00",
  "involt":11.75,
  "outcurr":"0.00",
  "outpower":"0.00",
  "outputEN":"1",
  "outvolt":"12.03",
  "state":"full_cv",
  "wh":"0.00",
  "prefs":{
    "adjustperiod":2000,
    "autosweep":600,
    "currentcap":8.500,
    "inPin":32,
    "lvProtect":"",
    "measperiod":200,
    "offthreshold":11.70,
    "pgain":0.005,
    "printperiod":1000,
    "psu":"drok",
    "pubperiod":1000,
    "ramplimit":2.000,
    "setpoint":0.000,
    "vadjust":27.00 
  }
}
```

and the serial logs:
```
MPPT 11.8Vin -> 0.00Wh PSU-out[11.99V 0.00A]-lim[14.71V 0.00A] ENABLED CLPS [pub-7]
MPPT 11.7Vin -> 0.00Wh PSU-out[11.99V 0.00A]-lim[14.71V 0.00A] ENABLED CLPS [pub-1]
MPPT 11.8Vin -> 0.00Wh PSU-out[11.99V 0.00A]-lim[14.71V 0.00A] ENABLED CLPS [pub-1]
MPPT 11.8Vin -> 0.00Wh PSU-out[11.99V 0.00A]-lim[14.71V 0.00A] ENABLED CLPS [pub-1]
```
(note the 0.00A current)

I tried to do a manual sweep:
```
SWEEP START c=0.000, (setpoint was 0.000)
First coming out of collapse-mode to clim of 0.00A
restore took 0.0s to reach 11.7V [goal 11.7], setting 0.0A
state change to sweeping (from mppt)
starting sweep
SWEEP DONE but zero un-collapsed points. aborting.
restore took 0.0s to reach 11.8V [goal 11.7], setting 0.0A
state change to mppt (from sweeping)
MPPT 11.8Vin -> 0.00Wh PSU-out[11.99V 0.00A]-lim[14.71V 0.00A] ENABLED CLPS [pub-2] COLLAPSED[4]
MPPT 11.8Vin -> 0.00Wh PSU-out[11.99V 0.00A]-lim[14.71V 0.00A] ENABLED CLPS [pub-8]
MPPT 11.8Vin -> 0.00Wh PSU-out[11.99V 0.00A]-lim[14.71V 0.00A] ENABLED CLPS [pub-7]
```

before that it said it was in constant current mode but with a current of 0.00A

I am not sure how it should work, the settings I manually set are:
- vadjust
- psu

if I disconnect the esp32 and set the drok output manually to 14.70V with a 2A limit (which is the charging limits written on this battery) I see some current flowing.

Did I miss something ?

schmurfy, 2020-11-11:
I tried with a lithium ion battery to check but I get the same result, if I set manually the voltage to 4.2 and the current limit to 0.2 there is current flowing (as chown by the drok) but when I connect the esp32 and do a sweep it sets the current to 0, here are the logs when I do a sweep:
```
SWEEP START c=0.000, (setpoint was 0.000)
state change to sweeping (from mppt)
starting sweep
SWEEPING 11.8Vin -> 0.01Wh PSU-out[2.11V 0.00A]-lim[4.20V 0.01A] ENABLED CLPS COLLAPSED[1] [adjusting 0.000A (from 0.012A)]
SWEEPING 11.8Vin -> 0.01Wh PSU-out[2.11V 0.00A]-lim[4.20V 0.01A] ENABLED CLPS
SWEEP DONE but zero un-collapsed points. aborting.
restore threshold now set to 11.73V
restore took 8.0s to reach 11.8V [goal 11.7], setting 0.0A
state change to mppt (from sweeping)
MPPT 11.8Vin -> 0.01Wh PSU-out[2.11V 0.00A]-lim[4.20V 0.00A] ENABLED CLPS COLLAPSED[2]
Starting AUTO-SWEEP (last run 0.1 mins ago)
SWEEP START c=0.000, (setpoint was 0.000)
state change to sweeping (from mppt)
SWEEP DONE but zero un-collapsed points. aborting.
restore took 0.0s to reach 11.7V [goal 11.7], setting 0.0A
state change to mppt (from sweeping)
MPPT 11.8Vin -> 0.01Wh PSU-out[2.11V 0.00A]-lim[4.20V 0.00A] ENABLED CLPS COLLAPSED[3]
MPPT 11.8Vin -> 0.01Wh PSU-out[2.11V 0.00A]-lim[4.20V 0.00A] ENABLED CLPS
```

there a long pause before `restore threshold`

t413, 2020-11-12:
Hey @schmurfy! Great to hear from you, sorry it’s not working right off the bat! 

It sounds to me like it’s a vadjust setting issue since it’s showing you an input voltage (voltage of the solar panel) as consistently 11.8. 

What have you got on the solar panel side? If you have a volt meter what voltage do the panels provide to the input of the Drok?

Also feel free to join in on the discord! Messaging makes debugging a little easier!

schmurfy, 2020-11-12:
@t413 Hello :)
Since sun is rather sparse currently I did my testing with a 12v power supply (I check with a volt meter and it is almost exactly 12v ), I tweaked `vadjust` to find the best match but I have to settle between 11.8 and 12.2 which are both wrong (27 & 28 respectively)

Ps: the discord invite link is invalid

t413, 2020-11-29:
Hey @schmurfy! I’ve gone and updated the discord links to non-expiring ones— I never realized expiring ones were default 🤦🏼‍♂️

Getting the vadjust close is all that matters, nice. 

So what battery or load is your power supply connected up to? Looks like the voltage output on it is set to 2.2V, that can’t be intended is it? You’ll need to set your power supply output voltage to the maximum your battery will charge to and the program will adjust the current to maximize the power it can get from the input. Set that voltage on the drok front panel or use the `outvolt=9` command. 

Very sorry for the late reply!

austrisv, 2021-03-14:
A lot of time has passed .. if this is still actual - isn't it a problem that @schmurfy is trying to use 12V PSU to charge an almost full sla battery? Thus, it would need to be 13.xV on drok output, but it receives only 12V in input?

schmurfy, 2021-03-17:
My testing method is probable flawed, I am waiting for the sun to come back to start my testing again but it will be a few more months ^^

closed 2025-04-25 by schmurfy