opensolarproject/OSPController #20: Adjust Charging for Temperature Changes

flywire, 2020-11-29:
See [ARDUINO PWM SOLAR CHARGE CONTROLLER ( V 2.02)](https://www.opengreenenergy.com/post/arduino-pwm-solar-charge-controller-v-2-02) Step 6.

>**Why Temperature monitoring is Required?**
>
>The battery’s chemical reactions change with temperature. As the battery gets warmer, the gassing increases. As the battery gets colder, it becomes more resistant to charging. Depending on how much the battery temperature varies, it is important to adjust the charging for temperature changes. So it is important to adjust charging to account for the temperature effects. The temperature sensor will measure the battery temperature, and the Solar Charge Controller uses this input to adjust the charge set point as required. The compensation value is - 5mv /degC/cell for lead-acid type batteries. (–30mV/ºC for 12V and 15mV/ºC for 6V battery).The negative sign of temperature compensation indicates an increase in temperature requires a reduction in charge set point. For more details, you may follow [this article](https://batteryuniversity.com/learn/article/charging_at_high_and_low_temperatures).

[image]
[image]

t413, 2020-11-29:
I’ve got a pack of these same sensors on my desk and a half-complete code change— but it’s fallen by the wayside of priorities sadly. Are you planning on running lead acid batteries stored outdoors? I’m happy to reprioritize if so

Great documentation reference!

flywire, 2020-11-30:
> Are you planning on running lead acid batteries stored outdoors?

Yes, in a vehicle.