opensolarproject/OSPController #21: Drock question

MikeGarber, 2020-12-23:
How does your PSupply survive, given https://www.droking.com/cs/support/topic/backdriving-the-200310-as-a-solar-battery-charger/.

Mine didnt.

t413, 2020-12-23:
Holy crap, it killed yours? I’m so sorry! I’ve had numerous numerous of these drok’s and that’s never been an issue. What exactly happened?

I do support running with a diode though. In that case it can’t read the battery voltage when it’s not charging which means you’d want another way of monitoring it. Support for energy monitoring is a big todo item for me

MikeGarber, 2020-12-23:
Been there, did that (if you read the DROCK thread).  I suspect it has to do with the leakage thru the Schottky diode I used (also asked in the thread but unanswered)

With a regular 5A power diode, there's no apparent current monitoring... https://www.droking.com/cs/support/topic/charging-battery-with-the-200310-current-monitoring/

t413, 2020-12-23:
Well it's very strange that your Drok died from being battery-side powered. I've had a drok running 24/7 for 1.5 years now being back powered at night and at least one other user around the world doing the same. 

If you'd like to try a different power supply option beyond Drok– I've now written and throughly tested support for [DPS5005](https://www.aliexpress.com/item/1000004290003.html) buck converters (only the ones that come with either USB or BT though, importantly).

MikeGarber, 2020-12-24:
Who knows?  They mightve changed something.

As to DPS5005.... modbus, eh?

I already ordered some hall effect current sensors & if I can continue to count on this PS for CV mode, I can FORCE it (essentially) into CI operation.    

But that DPS5005 looks WELL to look into!!
Thanx!

rin67630, 2021-02-15:
Backfeed kills the buck MOS-FET if there is enough current flowing backwards. 
The built in Schottky diode in the MOS-FETs aren't rated for much current, far less than the forward current. 
A small inadvertent short on the panel side and you are greeting the magic smoke.
You should operate your GROK with an ideal diode module:
- between the GROK and the battery if you want to minimize the GROK own consumption overnight, but you lose the battery monitoring
or
- between the GROK and the  panel if you want to keep the battery voltage monitoring, but you lose ~ 1 Ah overnight backfeeding the GROK.

rin67630, 2021-02-16:
I just checked a DPS5020 for it's quiescent currents at 14V
- forward from the panel 40mA that could have been acceptable, **but**
- backward reverse-fed from the battery 115mA!   and it feeds back to the input: a panel connected could add even more current
You really need an ideal diode module if you don't want to waste ~ 20Wh during a long winter night.

A DPS 5005 is less wasteful: it draws 40mA backfed. But that is still ~7Wh wasted when it's hard to get some juice from the scarce winter sun.