opensolarproject/OSPController #27: Not possible to compile / Serial Issue

liquid-light, 2022-05-08:
Hej

I wasn't able to send any command through the serial terminal. I get a lot of feedback from the ESP32S about whats wrong with the voltages and so on. But it doesn't accept any commands.

So I've tried to build the project in PlatformIO. The available binary seems to be outdated anyways.
Unfortunately this is also not possible.
I get this message :
`In file included from lib/MPPTLib/solar.h:2,
                 from src/main.cpp:1:
lib/MPPTLib/publishable.h:3:10: fatal error: FreeRTOS.h: No such file or directory`

But regardless of which version of FreeRTOS I try, it will never compile.

I think this is a great project and I would really like to help with improving the documentation. I think it's missing a lot of things for people who are not so deep involved in programming but more in electronics. But first I need to get my system up and running. ;)

My Setup is a ESP32S with a DPS5020

liquid-light, 2022-05-09:
After hours of searching on discord I found at least a workaround for my serial problem. After solving this it never accepted the `psu` command. I also found a workaround for this. It seems to be necessary to make a OTA update after flashing the latest version from Github. 🙄

From my perspective this should be either included in the documentation or, in the best case, not necessary.

The compile problem is still an issue! 🤷‍♀️

jonnycyborg, 2022-05-15:
@liquid-light just ran into this issue as well. Been putting off this project till the sun decided to come out.
Not far along, but was able to get it to compile by changing/adding this in [publishable.h](https://github.com/opensolarproject/OSPController/blob/c655a1dd8b6f5e1d9bd5f28c2bbf7fa416d788fc/lib/MPPTLib/publishable.h#L3)

This:
```
#include <FreeRTOS.h>
```
To This:
```
#include "freertos/FreeRTOS.h"
#include "freertos/semphr.h"
```

Not sure I fully grasp the details of the fix but can make a PR if it seems right!

liquid-light, 2022-05-15:
@jonnycyborg 
This solved the problem. Now it compiles! 👍 😀
Thank you!

jonnycyborg, 2022-05-15:
Of course! My mistake above, we don't need the quotes since it's not a user defined library. PR created!

hyfinn, 2023-04-24:
Thanks for posting this fix @jonnycyborg - did you have issues connecting to an MQTT server afterwards? I was using the .bin file for firmware before and was able to connect fine. But, since compiling locally it will connect to WiFi but not he server anymore - getting and error in serial: `[120037][E][WiFiGeneric.cpp:1476] hostByName(): DNS Failed for `

Any ideas?

PR #28 (2023-10-17): Location change of FreeRTOS
Fixes #27

Looks like things have changed for FreeRTOS and we just need to point to a different location.
Also needs an inclusion of semaphore now since its type definition doesn't seem to be included in the main library.
closed 2023-10-17 by t413