opensolarproject/OSPController #30: Unable to flash firmware

ways, 2023-02-25:
Hi

I really hope this project isn't dead.

> flash with esptool.py: esptool.py --chip esp32 --port /dev/ttyUSB0 write_flash 0

This will flash successfully, but monitoring the output only gives:

> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> invalid header: 0x04030000
> invalid header: 0x04030000
> invalid header: 0x04030000
> invalid header: 0x04030000
> invalid header: 0x04030000
> invalid header: 0x04030000
> invalid header: 0x04030000
> ets Jul 29 2019 12:21:46

(repeats)

The nodemcu-32 is brand new.

ways, 2023-03-03:
The idea seems so good I might rewrite this in esphome or micropython since I can't get this one to work.

Tobiasonline, 2023-03-10:
try compile the sourcecode with vscode and platformio and flash your own binary. That works for me

hyfinn, 2023-04-24:
> The idea seems so good I might rewrite this in esphome or micropython since I can't get this one to work.

Hey @ways - did you ever get round to rewriting? I'd be interested in any work you've done on this!

ways, 2023-04-25:
> > The idea seems so good I might rewrite this in esphome or micropython since I can't get this one to work.
> 
> Hey @ways - did you ever get round to rewriting? I'd be interested in any work you've done on this!

Yes, I have it in a private repo. Will push to gitlab and post link within a few days.